bigtreetech/BIGTREETECH-TMC5160-V1.0 #2: Can't get SKR 1.3 + TMC 5160 to work

extesy, 2019-08-19:
I'm having trouble connecting TMC 5160s to SKR 1.3 board. When I power up the board I see "3d printer ready" message for 2-3 seconds which then changes to "driver error, printer halted, please reset". The same five 5160s were working fine with another board (MKS GEN L) but they don't work on SKR 1.3.

Attaching config files: [configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3513829/configuration.zip)

When debugging with MONITOR_DRIVER_STATUS I've noticed that the drivers randomly trigger otpw warnings despite not being actually connected to any steppers. Every time I run M122 this warning is displayed for a random set of steppers (I'm using X+Y+Z+E0+E1).

Cross-posting from https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/69

extesy, 2019-08-19:
@Msq001 @bigtreetech Can you comment please? Is this a hardware problem with 5160s?

bigtreetech, 2019-08-28:
found the reason, just need to break the third pin from the right, because of this pin is the CLK of the chip, but the SCK of the spi have be wired with this pin by motherboard, This causes the main frequency of the chip to be provided by SPI SCK, So all you need to do is modify as following figure. and tmc5161 is the same reason.
[image]

bigtreetech, 2019-08-29:
TMC5160_V1.2 is the same
This pin is an external clock pin, which is mostly wired to GND，that the main frequency is provided by the internal clock of the chip, So this pin is usually unnecessary. 
I test it on SKR V1.3, Destroy this pin will not affect the normal use of other motherboard. 
In fact, MKS GEN_L must need destroy this pin too. because of it wired spi-sck with clk too.
[image]

silentcleric, 2019-08-29:
I can confirm this works with the V1.2 with the SKR 1.3 . I was also able to revert back from the code fix that was provided-[ Fix support for BIGTECHTREE SKR 1.3 board teemuatlut/TMCStepper#63](https://github.com/teemuatlut/TMCStepper/pull/63/commits/c481cccf542f3243c2c0594f5cf765fde2dd22d0)

shreeramlive, 2019-11-03:
Yes I can also confirm this. So far this was the only fix which I could find to get **TMC5160 SPI** drivers to work with SKR V1.3. [Fix for TMC5160](https://github.com/teemuatlut/TMCStepper/pull/63/commits/c481cccf542f3243c2c0594f5cf765fde2dd22d0) Thanks for fixing this issue @extesy and confirming @silentcleric

closed 2020-01-14 by bigtreetech

iniciando-66, 2020-04-28:
Greetings
I'm an electronics technician working in factory maintenance here in Brazil and I would like to ask my friends about the Bigtreeteck TMC5160 v1.2 drive; I have been using the A4988 and TB6600 to control 1 Nema stepper motor controlled by a PWM signal generator board, but I would like to improve all of this by migrating to this new platform; I set up a prototype using the arduino uno to generate the spi signals but there is no reaction from the engine, do I need to install some other program before, or is it ready for use? Observing with the microscope the pins 20 (Vcc_IO) 21 (SD_MODE) and 22 (SPI_MODE) are all connected together in the IC, in this version there is no possibility to configure the operation in standalone (pin SPI_MODE low and SD_MODE high)?
 Thank you very much for your attention.

Alaskit, 2020-07-15:
Hello, 
Please help my. I have maybe good FW for BTT SKR v1.4 Turbo, but my print is not work. 
I have BTT SKR v1.4 Turbo and TMC5160 v1.5. My old Printer is Tronxy X5SA, but is not good.. However, a new upgrade will resolve this issue. My thermometer HotBed is ok and HotEnd is ok this all perfect. But command, Repetier or Prontface not work for run my Printer. I testing 4x FW, but Printer not work. Cooling Fan not work, Run for X / Y / Z+Z1 / E0 not work.  I'm afraid to try heating. My FW have activation sensors, serv's, i set for other items. Upload FW i ok, but only thermistors work.  
I don't know any people who can advise me and google doesn't help me. Unfortunately, even my English is bad.
I don't know what pins I should have on the driver, or whether I should modify them. That's why I tried everything ... bend the third pin, connect GND and DC0, on the motherboard jumper's for UART or SPI ... nothing helps.
Please help, this problem has been around for 14 days .. and I don't know what to do next. I'm sending files here, maybe they can help me. 

[My FW Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/files/4928346/My.FW.Marlin.zip)