bigtreetech/BIGTREETECH-TMC5160-V1.0 #4: TMC5160-V1.2

dineshvyas, 2019-12-10:
kindly release documments for TMC5160-v1.2. My drives just stopped working.

DarioX7, 2020-01-19:
Please release the documentation to version V1.2

Quinnthouzand, 2020-06-17:
BTT TMC5160s not working correctly on my SKR 1.4 Turbo and I can't seem to find any solid resource for configuration.

evandene, 2020-06-28:
> BTT TMC5160s not working correctly on my SKR 1.4 Turbo and I can't seem to find any solid resource for configuration.

I desperately looking for help to let work my SKR1.4 with 5x stepper driver tmc5160
I have no idea what's going on
Marlin 2.0
Delta Printer, SKR1.4
BLTOUCH

davidmoreland, 2021-01-12:
what steppers are you driving?  After weeks of research and trial/error.  The only way I got my GTR board / BTT TMC 5160's to work correctly was to drive 2 motors in series...  Must be an impedance issue, but haven't investigated farther.

evandene, 2021-01-12:
I gave up,
Scrapped the board

Ed van den Enden

filipembedded, 2023-08-01:
> Please release the documentation to version V1.2

Have you found the documentation for TMC5160-V1.2. I searched all over Internet, couldn't find a thing, just found the documentation for TMC5160-V1.0 and TMC5160-V1.1. How do they expect for people to use this product, maybe with some witchcraft..