bigtreetech/BIGTREETECH-TMC5160-V1.0 #5: Not working SPI: SKR 1.3 + TMC5160 v1.2 + Marlin 2.0 (5103cdc)

bmgoncu, 2019-12-22:
Hi,

I purchased the SKR1.3 and TMC5160 v1.2 from BigTreeTech, i first tried to simply connect the driver to the board but the command M122 returned  Testing X connection... Error: All LOW on all axises.
Then i tried to bend out the CLK pin on the driver and tried again and still got an error, but this time it reported Testing X connection... Error: All HIGH (Full error output and my configs are below).
Please assist me.

[TMC5160 SKR 1.3 spi m122 error.txt](https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/files/3992362/TMC5160.SKR.1.3.spi.m122.error.txt)
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/files/3992364/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/files/3992365/Configuration_adv.txt)

shreeramlive, 2019-12-22:
There is already a **fix** for this issue. You just need to tweak some code in the Marlin firmware.
Read the thread here #2 . @extesy has provided a fix https://github.com/teemuatlut/TMCStepper/pull/63/commits/c481cccf542f3243c2c0594f5cf765fde2dd22d0.
For this to work you also need to enable SW SPI in Configuration_adv.txt
Good luck!

bmgoncu, 2019-12-22:
Modified the code in marlin under the .pio folder as the commit diff specified but i still get the error as attached.
(Please note i tried it with the CLK pin being disabled)
[TMC5160 SKR 1.3 spi m122 error2.txt](https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/files/3992460/TMC5160.SKR.1.3.spi.m122.error2.txt)

shreeramlive, 2019-12-23:
That fix worked for me without any hardware modifications, I didn't bend any pins.
Try without bending the **CLK pins**.

Also enable `TMC_USE_SW_SPI `in **Configuration_adv.txt**

bmgoncu, 2019-12-23:
The TMC_USE_SW_SPI  is already enabled as you say;

`  #define TMC_USE_SW_SPI`
`  //#define TMC_SW_MOSI       -1`
`  //#define TMC_SW_MISO       -1`
`  //#define TMC_SW_SCK        -1`

I tried as you say with the CLK pin connected, The error is now 

> Testing X connection... Error: All LOW
The detailed output of M122 is below. 
[Error with clk connected.txt](https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/files/3995816/Error.with.clk.connected.txt)

Besides that can the TMC5160 drivers work with usb 5V power? (for diagnostig etc, not to drive the actual motors...)

extesy, 2019-12-23:
@bmgoncu You always need to provide external power when drivers are installed. No communication with drivers will work with USB-only power.

bmgoncu, 2019-12-29:
@extesy yes you were right sorry that i missed such a trivial point, the m122 now returns all drivers to be ok. İ cant move the axis for some reason right now but i suspect that it is something to do with the voltages or steps/mm or something else

bmgoncu, 2019-12-29:
Ok, after some digging around i found the cause of it. İt seems like the drivers are not initialized properly by marli  on startup. The motors dont move when i try to move x and y on my corexy machine but when i load from eeprom (m501) and save (m500) it again i can get one axis to move and the other moves very little. İs this issue known (if so can you point me to the issue page) and related to marlin or the tmc driver?

wickedsun, 2020-01-13:
I am in the same boat it seems, but after a lot of research and a lot of fiddling around, if I spam M122, all of my driver eventually say "Bad response", and ALL LOW". They do not stay that way, they cycle.

My steps are constantly wrong, after a few M500/M501 eventually they are in a "good" state. They will keep cycling through a bunch of registers and eventually Bad Response. I've tried removing the CLK pin. I've reflashed countless times, I've tried Marlin 2.0 bugfix, BIGTREETECH's version. I always end up with the same result. I have SW_SPI enabled.

I do not know what could be wrong.

`22:39:50.131 : Driver registers:22:39:50.131 : X	0xC0:00:00:0022:39:50.131 : Y	0x01:0B:00:0022:39:50.131 : Z	0x80:0B:40:0022:39:50.131 : E	0x01:05:00:0022:39:50.131 : Testing X connection... OK22:39:50.132 : Testing Y connection... OK22:39:50.132 : Testing Z connection... OK22:39:50.132 : Testing E connection... OK
----
22:40:49.134 : Driver registers:22:40:49.134 : X	0x00:0B:40:0022:40:49.134 : Y	0x01:0B:00:0022:40:49.135 : Z	0x00:00:00:00	 Bad response!22:40:49.135 : E	0x01:05:00:0022:40:49.135 : Testing X connection... OK22:40:49.135 : Testing Y connection... OK22:40:49.135 : Testing Z connection... OK22:40:49.135 : Testing E connection... OK
----
22:41:17.224 : Driver registers:22:41:17.224 : X	0x00:0B:40:0022:41:17.224 : Y	0x00:00:00:00	 Bad response!22:41:17.224 : Z	0x80:00:00:0022:41:17.224 : E	0x80:00:00:0022:41:17.224 : Testing X connection... OK22:41:17.224 : Testing Y connection... Error: All LOW22:41:17.224 : Testing Z connection... OK22:41:17.224 : Testing E connection... OK
`

bmgoncu, 2020-01-13:
Hi,

I figured out a solution, make the above mentioned code fixes for the TMCSTEPPER library and configure your r-sence values and current values correctly(see my configuration adv file) after that you **must push the drivers CLK pin up** using a soldering iron and **solder a jumper wire between the drivers CLK and GND pin**. The problem seems that the onboard Clock signal interferes with the drivers and causes all sort of ruckus.

closed 2020-01-13 by bmgoncu

wickedsun, 2020-01-13:
Is this the same on a SKR1.4 ?

bmgoncu, 2020-01-13:
Dont have the hardware to test it on but i suspect yes, its a problem on the driver side

wickedsun, 2020-01-13:
Thank you very much for your answer.

So much for an easy SW SPI board...

wickedsun, 2020-01-13:
Well quick report back in for other people experiencing the same thing. Just changing the code seemed to have worked on the SKR1.4 Turbo. I believe they pull down on the CLK pin on the revised design, so no need to ground the CLK pin anymore.

Now to the test print.

evandene, 2020-06-24:
I have installed the TMC5160 drivers on my SKR1.4 board but I do still get the ALL LOW error
Can somebody help me please? I attached the config files.
For your information I'm using the board on a Delta Printer
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/files/4823717/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/files/4823718/Configuration_adv.txt)