bigtreetech/BIGTREETECH-TMC5160-V1.0 #8: Missing Documentation 1.2 / Motion Controller Mode

AlpineWhite, 2020-01-19:
Checking the documentation, there is none for version 1.2.  The motion controller mode also appears selectable via solder pads on version 1.0, but there is no reference as to where those pads are on the chip.

Can you please update the documentation for those of us who wish to use the motion controller mode?

PMTaylor, 2020-01-26:
Agreed! I'm looking for this information.
Any leads @AlpineWhite ?

PKav89, 2020-02-25:
Motion controller mode is not possible on v1.2. Pins 20, 21 and 22 (VCC_IO, SD_MODE and SPI_MODE) are tied together on the PCB. It is possible to cut this VERY SMALL trace and make a jumper wire from SD_MODE to GND, but it's rather difficult.

PMTaylor, 2020-02-26:
Yeah, i'd figured this out by myself. Schematics on the site would be very helpful though, irrespective of the answer to this question.
If you do a V1.3, please return a solder bridge to enable disable internal motion control (like v1 scematics). It's one of the nice features of this chip, and it's a tragedy to make it impossible to use, even if 3d printers are currently too dumb to use the feature.

PKav89, 2020-02-26:
Well, according to a Watterott schematic https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC5160_v14.pdf there is no pulldown resistor, so it seems to be possible to switch TMC5160 into motion controller just by cutting SD_MODE trace and leaving SD_MODE pin floating. I'll test it later.

PKav89, 2020-02-26:
I just tried cutting SD_MODE trace leaving SD_MODE floating and it seems to work.

IOIN before cut (tied to VCC_IO) 0x30:00:00:52
IOIN	after cut (SD_MODE floating) 0x30:00:00:12

So, bit 6 in the IOIN register changed from 1 to 0, which, according to datasheet, is "SD_MODE (1=External step and dir source)". I will try motion controller mode soon.

PKav89, 2020-02-27:
I just tested motion controller, and yes, it runs OK with SD_MODE pin floating. Just cut the small trace on the board and TMC5160 switches into motion controller mode.

madhunm, 2020-04-09:
> I just tested motion controller, and yes, it runs OK with SD_MODE pin floating. Just cut the small trace on the board and TMC5160 switches into motion controller mode.

Can you please post pictures of which trace you cut?

iniciando-66, 2020-04-29:
Greetings
I'm an electronics technician working in factory maintenance here in Brazil and I would like to ask my friends about the Bigtreeteck TMC5160 v1.2 drive; I have been using the A4988 and TB6600 to control 1 Nema stepper motor controlled by a PWM signal generator board, but I would like to improve all of this by migrating to this new platform; I set up a prototype using the arduino uno to generate the spi signals but there is no reaction from the engine, do I need to install some other program before, or is it ready for use? Observing with the microscope the pins 20 (Vcc_IO) 21 (SD_MODE) and 22 (SPI_MODE) are all connected together in the IC, in this version there is no possibility to configure the operation in standalone (pin SPI_MODE low and SD_MODE high)?
Thank you very much for your attention.

PKav89, 2020-04-29:
I wrote about it one post back. Just cut the SD_MODE trace and driver switches to Motion Control mode.

madhunm, 2020-04-29:
> > I just tested motion controller, and yes, it runs OK with SD_MODE pin floating. Just cut the small trace on the board and TMC5160 switches into motion controller mode.
> 
> Can you please post pictures of which trace you cut?

Please share pictures

PKav89, 2020-04-29:
I don't have a good camera to take a photo of such small things.
https://imgur.com/0LkeWHb
This tiny trace, the middle one.

xukangmin, 2023-01-04:
On V1.3, PIN 23 SD_MODE on the TMC5160-WA is connected to the solder jumper on the top of the board. The resistor is connected to VCCIO by default. You can jump the other way, then it's connected to GND. See image below. 

Hope it helps anyone searching for solutions.

[image]

psychuk, 2023-07-01:
@xukangmin just bought some of these and want them to work with SPI.
Could you tell us what jumpers need to be connected within your green circle?

xukangmin, 2023-07-02:
You can either remove the resistor or solder the resistor to the other way.
Both should work.